cybergalactic/MSS #1: About the motion response ampliotude operators

Ices1995, 2019-04-28:
Thanks for the convenience you offerd.
I'm a doctoral student and I'm reading a book written by your team - 'Ship Motion Control'.
In chapter 4, the book mentioned 'Motion RAO'  were given as a function of the wave frequency in the toolbox,  I want to study RAO by the function but I can't find it.
Do I omit it or it has been removed ?

cybergalactic, 2019-05-01:
You find the motion RAOs for several vessel stored as Matlab mat files. They are located under /HYDRO/vessels_wamit and /HYDRO/vessels_shipx. You can load these data files into the workspace. Try using 

> load supply
> load tanker

Then you find the RAOs in the vessel structure by simply typing 
>vessel

closed 2020-03-30 by cybergalactic

NimaJim, 2024-06-06:
Thank you, @cybergalactic, for sharing your great work. I am trying to simulate the wave-induced heave motion of a semi-submersible rig. I wonder if MSS is able to compute RAOs for any arbitrary semi-submersible vessel it just uses the RAOs provided by WAMIT, ShipX as an input?
If there is no way to use MSS to calculate RAOs, can you please suggest references that can be efficiently used to do that?

cybergalactic, 2024-06-06:
You are correct. The RAOs must be computed using a hydrodynamic code. It cannot be done in MSS. In MSS, two scripts, 'wamit2vessel.m' and 'verses2vessel.m', read the output files of WAMIT and ShipX. If you have a different program, for instance, an open-source university code, you can compute the RAOs with this code. Then, you need to modify the MSS files to import your data to the workspace and use the Simulink templates to simulate your vessel in waves. The Simulink templates assume that the vessel data has a certain structure as specified in the configuration guide for ShipX and WAMIT. In particular, all hydrodynamic computations are done for incoming waves at 0:10:180 degrees. This allows for interpolation between the data in Matlab/Simulink. For more details, you can read the configuration guide:

https://github.com/cybergalactic/MSS/tree/master/documentation/Configuration%20guide%20for%20ShipX%20and%20Wamit%20vessels%20models

NimaJim, 2024-06-06:
Thank you so much for your super fast response.