cybergalactic/MSS #12: Reference request for navigation files.

wkyouncnu, 2021-04-10:
Thank you for sharing the great resources. 

Is there any paper or books for the following navigation filters, including ins_ahrs, ins_euler, ins_mekf, and ins_mekf_psi?

I would like to know the detail about the Jacobian matrix of the aforementioned filters.

- Navigation (m-files)
- acc2rollpitch static roll and pitch angles from specific force
- gravity acceleration of gravity as a function of latitude using the WGS-84 ellipsoid parameters
- ins_ahrs Error-state Kalman filter for INS aided by position and AHRS measurements
- ins_euler Error-state Kalman filter for INS aided by position and yaw angle measurements
- ins_mekf Error-state Kalman filter for INS aided by position and magnetic field measurements
- ins_mekf_psi Error-state Kalman filter for INS aided by position and yaw angle measurements
- insSignal basic INS signal generator

cybergalactic, 2021-04-10:
The theory for the error-state Kalman filter (unit quaternions and Euler angles representations) is quite complicated. I have written a textbook chapter on aided inertial navigation systems (INS) explaining this in detail. Consequently, the m-files and the methods are found in Chapter 14 in my 2021 Wiley-Blackwell textbook:

https://www.wiley.com/en-ir/Handbook+of+Marine+Craft+Hydrodynamics+and+Motion+Control%2C+2nd+Edition-p-9781119575054

wkyouncnu, 2021-04-10:
@cybergalactic 

Thank you for the response.

Your **mssSimulink** file seems to work only in MATLAB 2021, but I am currently using MATLAB 2017b only,

thus I was not able to open and run the **mssSimulink** file

Can you  save the above-mentioned files in prevision version such as Matlab 2017b so that I can open it?

In order to do that,

- In  the Simulink Editor, on the Simulation tab, select Save > Previous Version.  
- In the Export to Previous Version.

Thank you in advance.

cybergalactic, 2021-04-15:
We cannot take the time to save the repository in previous Simulink versions since there are too many versions. Backward compatibility is also a problem.  In addition, the library file mssSimulink.slx is not the only SImulink file that is needed to run the MSS examples.

closed 2021-04-15 by cybergalactic