cybergalactic/MSS #15: Question regarding reference for compass linearization

mnissov, 2021-05-30:
In `ins_mekf_psi` lines 79-86:
```matlab
% Linearization of compass measurement
a = (2/q_ins(1)) * [q_ins(2) q_ins(3) q_ins(4)]'; % 2 x Gibbs vector
u = 2 * ( a(1)*a(2) + 2*a(3) ) / ( 4 + a(1)^2 - a(2)^2 - a(3)^2 );
du = 1 / (1 + u^2);
c_psi = du * (1 / ( 4 + a(1)^2 - a(2)^2 - a(3)^2 )^2 ) * ...
        [ -2*((a(1)^2 + a(3)^2 - 4)*a(2) + a(2)^3 + 4*a(1)*a(3)) 
           2*((a(2)^2 - a(3)^2 + 4)*a(1) + a(1)^3 + 4*a(2)*a(3)) 
           4*((a(3)^2 + a(1)*a(2)*a(3) + a(1)^2 - a(2)^2 + 4)) ];  
```

Do you have a reference for this transformation to the "2x Gibbs vector"? I looked through "Handbook of Marine Craft Hydrodynamics and Motion Control" but couldn't seem to find anything in the quaternion sections.

cybergalactic, 2021-05-30:
There is no reference for this equation. I was not able to find anything on Internet so I derived the equation from scratch by using the rotation matrix elements R21 and R11 parametrized in terms of Gibbs parameters,  see Equation (14.250), which did the trick.  An excellent reference for Gibbs parameters and kinematics in general is the textbook: "Space Vehicle Dynamics and Control",  2nd edition 2008,  by Bong We (https://arc.aiaa.org/doi/10.2514/4.860119). Here you will find Section 5.4.2 very useful.

The "2 x Gibbs vector" parametrization is used to obtain physical quantities in the Kalman filter covariance matrix. It is briefly mentioned below Equation (14.228)  in my textbook where it is stated:" It is convenient to scale Gibbs vector by a factor 2 such that the KF covariance estimates will be given in radians squared, which is equivalent to angle errors using a first-order approximation".  This is just something we do as an implementation trick and I do not have a reference for it. My motivation for writing the INS chapter and m-file functions was that there are many tricks and many of them are not published.

mnissov, 2021-05-30:
I'm assuming equations 14.250 and 14.228 are in the newer edition of "Handbook of Marine Craft Hydrodynamics and Motion Control", at least I can't seem to find them in my version.

cybergalactic, 2021-05-30:
There is no INS chapter in the first edition of the book from 2011. All the INS algorithms are in the second edition of the book published April 2021. The new Chapter 14 is entitled "Inertial Navigation Systems".

mnissov, 2021-05-30:
Okay, looks like I'll have to source a copy of the new edition then. Thanks for the help and for the Bong Wie reference.

closed 2021-05-30 by cybergalactic