cybergalactic/MSS #16: run "demoOtherUSVHeadingControl" error,maybe exist problem

openye, 2021-07-09:
[image]

cybergalactic, 2021-07-09:
The Simulink file is running on my computer. Can you double check that you have downloaded the latest version of the m-files and Simulink files from GitHub. Also double check that all MSS files are on your Matlab path. The otter.m file should return xdot and U. You can test this using the call Matlab command: [xdot,U] = otter(zeros(12,1),[0 0]',1,[0 0 0]',1,1). Then report back.

openye, 2021-07-12:
My version of MATLAB is 2021a. After I add an external declaration, the program can be executed. Thank you very much, Professor fossen.
[image]
[image]

cybergalactic, 2021-07-12:
You are welcome.

closed 2021-07-12 by cybergalactic