cybergalactic/MSS #23: m2c.m matrix shape error

chihaya-kawamura, 2022-04-18:
from line 37 to line 43
```
M21 = M12';
M22 = M(4:6,4:6);

nu1 = nu(1:3);
nu2 = nu(4:6);
dt_dnu1 = M11*nu1 + M12*nu2;
dt_dnu2 = M21*nu1 + M22*nu2;
```

＞＞＞ I think it should be 
```
M21 = M12.';
M22 = M(4:6,4:6);

nu1 = nu(1:3);
nu2 = nu(4:6);
dt_dnu1 = M11*nu1.' + M12*nu2.';
dt_dnu2 = M21*nu1.' + M22*nu2.';
```

cybergalactic, 2022-04-18:
MATLAB uses the apostrophe operator ( ' ) to perform a complex conjugate transpose, and the dot-apostrophe operator ( . ' ) to transpose without conjugation. For vectors/matrices containing all real elements, the two operators return the same result. Hence, this will not affect the results since M11, M12, M21, M22, nu1 and nu2 are all real elements.

chihaya-kawamura, 2022-04-18:
Sorry, I made a mistake.
Thank you Professor.

closed 2022-04-18 by cybergalactic

wtllll, 2022-10-11:
您好，已收到您的邮件，谢谢！