cybergalactic/MSS #24: Question: Hull lifting forces

gabrielfpacheco, 2022-04-18:
Hi @cybergalactic,

I'm working on a 6-DOF simulation model for a flatfish-shaped underwater vehicle and using your textbook as a reference. I'm able to obtain fair estimates for the added mass matrix coefficients using hydrodynamic software and I can also determine rough estimates for linear viscous damping coefficients based on existing data of similar vehicles - being an underwater vehicle, linear potential damping is being neglected. To determine non-linear damping effects though, I'm using surge resistance and cross-flow drag principles.

However, as far as I could understand, in many modeling approximations of `D(v) = D + Dn(v)` the effects of lift on the vehicle's hull are not explicitly accounted for. From your textbook:

> **Lifting Forces**: Hydrodynamic lift forces arise from two physical mechanisms. The first is due to the linear circulation of water around the hull. The second is a nonlinear effect, commonly called cross-flow drag, which acts from a momentum transfer from the body to the fluid. This secondary effect is closely linked to vortex shedding.

What I mean is that I don't see where the lift generated from the hull moving in the fluid with a certain angle of attack being considered in the models, and I would assume that they play an important role for survey AUVs (moving the vehicle up-down). To my understanding, the elevators would be responsible for generating a pitch moment, but the most representative lift would be generated by the hull; which would have a way bigger area for a flatfish-shaped AUV.

The following line is the only reference I could find so far that actually models the vehicle's hull as a wing profile and consider its lift-drag as a function of the vehicle's angle of attack.

https://github.com/cybergalactic/MSS/blob/d03cfc3cf9ba72600490795c45ac8c6dc61ece2d/VESSELS/remus100.m#L132

*Q: Is my understanding correct thus far? If so, why are aoa-dependent lifting forces not being considered using [this](https://github.com/gabrielfpacheco/MSS/blob/master/GNC/forceLiftDrag.m) for other vessels? Are these effects being accounted for elsewhere in other vessels models, such as in the [NPS AUV](https://github.com/cybergalactic/MSS/blob/master/VESSELS/npsauv.m) which does consider [cross-flow drag](https://github.com/cybergalactic/MSS/blob/master/VESSELS/npsauv.m#L160)?

I really appreciate the help in advance!

cybergalactic, 2022-04-19:
For most underwater vehicles the angle of attack is assumed to be small. This implies that lift is completely neglected and only drag is considered. This is also the key assumption for conventional surface ships. This is of course not true if you have "flying" underwater vehicles operating at higher speeds. Lift is only included for the Remus 100 AUV in the toolbox. You can of course add lift/drag to your flatfish-shaped UAV following the same procedure. Maybe plot the forces as a function of your operating speeds/angle of attack to check if the lift force is important or not. The NPS AUV model is based on a Taylor-series expansion of hydrodynamic derivatives. This is basically curve fitting. Since the model includes second-order velocity terms, lift and drag will be included.

gabrielfpacheco, 2022-04-20:
Thank you so much for the fast reply, Dr. Fossen.

> For most underwater vehicles the angle of attack is assumed to be small. This implies that lift is completely neglected and only drag is considered. This is also the key assumption for conventional surface ships.

Got it, thank you for clarifying.

> The NPS AUV model is based on a Taylor-series expansion of hydrodynamic derivatives. This is basically curve fitting. Since the model includes second-order velocity terms, lift and drag will be included.

Understood. But in that case, why does the model have a [cross-flow drag](https://github.com/cybergalactic/MSS/blob/master/VESSELS/npsauv.m#L160) calculation added to `Y, Z, K, M` components? Shouldn't they be already accounted for by these second-order velocity terms as well?

gabrielfpacheco, 2022-04-20:
Thanks a lot for your help! 

>  My best guess is that they have modeled hull damping by the cross-flow drag method and added the remaining second-order terms for appendices etc

That makes a lot of sense to me as a possible reason.

> You can also check the paper (or contact the authors)

I've looked into the paper earlier and couldn't find an explanation there, only a mention that it is considered as follows:

> The model described includes a model of the cross flow drag effects, and a model of the propulsion system and is therefore a large departure from the original work of Abkowitz. 

I'll try to reach the authors as you suggested.
Once again, I really appreciate the support, Dr. Fossen. I'm closing the issue

closed 2022-04-20 by gabrielfpacheco