cybergalactic/MSS #25: Discrepency vs. book

dc5048, 2022-06-01:
https://github.com/cybergalactic/MSS/blob/263d3fd14149cf5a82efb934d8b512b2a06acd4d/GNC/llh2flat.m#L22

Hello again Dr. Fossen,

There is a discrepancy between this line of code and equation 2.112 in your _Handbook_ (2nd edition 2021).  The latter includes a square-root around the denominator term. I don't have access to the cited reference, (Farrell 2008), but am thinking this may be a typo in the book while the MSS code is correct (in which case, might be useful to note in the function documentation). 

Compare to similar algorithm here: https://www.mathworks.com/help/aeroblks/llatoflatearth.html.

wtllll, 2022-06-01:
您好，已收到您的邮件，谢谢！

cybergalactic, 2022-06-01:
Thanks for letting me know.  The MSS code is the correct one, while the square root in the denominator of Equation (2.112) in the book should be removed.  Since the book is digitally produced, John Wiley will correct this in future prints when I notify them. For older prints, I will add the typo in the errata, see (https://www.fossen.biz/wiley/)

closed 2022-06-01 by cybergalactic