cybergalactic/MSS #29: A request for a reference paper

wtllll, 2022-07-29:
Dear professor Fossen. I'm a Phd student. I'm really interested in the project. But I can't find any website that can download the reference paper "Van Berlekom, W.B. and Goddard, T.A. (1972). Maneuvering of Large Tankers, Tansaction of SNAME, 80:264-298" in tanker.m. Could you please provide me any website that can download this paper?
Anyway, thanks a lot !

cybergalactic, 2022-07-29:
The paper is very old and hard to find. I found my copy in the library 30 years ago. 
[1972 BerlekomGoddard - Maneuvering of large tankers.pdf](https://github.com/cybergalactic/MSS/files/9218090/1972.BerlekomGoddard.-.Maneuvering.of.large.tankers.pdf)

wtllll, 2022-07-29:
Great！ Thanks !

closed 2022-07-29 by cybergalactic