cybergalactic/MSS #30: wamit2vessel.m mirrors RAO incorrectly to 190 to 350

F-Disk, 2022-12-16:
Dear Professor,

This issue is related to: **wamit2vessel.m**

WAMIT deliveres RAOs from 0 to 180 deg. I noticed that the motionRAO, forceRAO en driftfrcRAOs are mirrored incorrectly to 190 to 350. 
The heading, sign and phase are all incorrect in some cases.

I solved this by the following code from line 669:
%-------------------------------------------------------------------------%
%% map data from 0-180 deg to 180-360 deg (symmetry about x-axes)
%-------------------------------------------------------------------------
vessel.headings = (pi/180)*[0:10:350];

% To change sign phase is shifted 180deg
rao_sign = [0 1 1 1 0 0];
for i = 1:6
   vessel.motionRAO.amp{i}(:,20:36,1)   = vessel.motionRAO.amp{i}(:,flip(2:18),1);  
   vessel.motionRAO.phase{i}(:,20:36,1) = rao_sign(i)*pi+vessel.motionRAO.phase{i}(:,flip(2:18),1);    
   vessel.forceRAO.amp{i}(:,20:36,1)    = vessel.forceRAO.amp{i}(:,flip(2:18),1); 
   vessel.forceRAO.phase{i}(:,20:36,1)  = rao_sign(i)*pi+vessel.forceRAO.phase{i}(:,flip(2:18),1);   
end      

rao_sign_WD = [1 -1 -1];
for i = 1:3
   vessel.driftfrc.amp{i}(:,20:36,1)   = rao_sign_WD(i)*vessel.driftfrc.amp{i}(:,flip(2:18),1);     
end 
clear rao_sign_WD

wtllll, 2022-12-16:
您好，已收到您的邮件，谢谢！

cybergalactic, 2022-12-19:
Thanks for sharing the mirroring bug. The flip function solves this elegantly. Could you please explain the following sign corrections: 

1) rao_sign = [0 1 1 1 0 0] changes the phases of the flipped data with +pi in sway, heave and roll (DOFs 2, 3, 4) while surge, pitch and yaw (DOFs 1, 5, 6) are unchanged
2) rao_sign_WD = [1 -1 -1] changes the phases of the flipped data with +pi, -pi, -pi in surge, sway and yaw. Note that under item 1 the signs are  [0 1 0] if you consider elements 1,2 and 6.

Once this is sorted out, I will update the code on GitHub.

F-Disk, 2022-12-20:
Thank you!

forceRAO and motionRAO have amp and phase. driftrc has only amp.

1. To change the sign of forceRAO and motionRAO, one can shift the phase with pi (180 degrees). Also one could change the sign of the amplitude, but I preferred to have all amplitudes with positive sign.
I do think it is incorrect to change the sign of the phase of the RAO, like it was implemented in wamit2vesssel.m.
Because the RAO is mirrored in the X-axis:
Surge(10deg) = Surge(190deg)
Sway(10deg) = -Sway(190deg)
Heave(10deg) = Heave(190deg)
Roll(10deg) = -Roll(190deg)
Pitch(10deg) = Pitch(190deg)
Yaw(10deg) = -Yaw(190deg)
So this leads to phase shift for sway, roll and yaw -> rao_sign should be [0 1 0 1 0 1], and not [0 1 1 1 0 0] like I mentioned earlier. 
Does this make sense?
This also rectifies your point in your comment "Note that under item 1 the signs are [0 1 0] if you consider elements 1,2 and 6"

2. To change the sign of driftfrc, the amplitude is changed in sign. Not like you mentioned flipped with pi in your comment.
Because the RAO is mirrored in the X-axis:
Surge(10deg) = Surge(190deg)
Sway(10deg) = -Sway(190deg)
Yaw(10deg) = -Yaw(190deg)

Best Regards

cybergalactic, 2022-12-20:
Yes, this makes sense. I will update wamit2vessel.m on GitHub. Again, thanks for sharing the bug fix for mirroring the WAMIT data to 180-360 degrees.

closed 2022-12-20 by cybergalactic