cybergalactic/MSS #32: Question regarding ExPathGen.m

joaobrittoneto, 2023-01-13:
Hi Prof. Fossen, 
I'm following the example ExPathGen.m (described in Example 10.4 of the 1st edition of the handbook).
I wonder how to use the formulation to perform a deceleration trajectory, i.e., having initial speed [U=5](https://github.com/cybergalactic/MSS/blob/master/mssExamples/ExPathGen.m#L238) at waypoint 0 (th=0) and Uref = 0 at waypoint 1 (th=1).
The issue I'm having is determining the parameter `th`, at which the deceleration starts.
Any guidance on how to properly formulate this case?

cybergalactic, 2023-01-16:
I do not have a quick fix. This usually involves solving a set of equations that need to be derived based on your requirements.

closed 2023-01-23 by cybergalactic

joaobrittoneto, 2023-01-24:
Just to register, I found a way to determine the deceleration trajectory by using the same formulation but moving backward along the path (from th=1 to th=0) doing a normal acceleration profile. At the end just flip the resultant vector to have the deceleration trajectory.

cybergalactic, 2023-01-24:
That makes sense. Often the simplest solution is the best one.

Thor