cybergalactic/MSS #33: Find some questions in the reference book

wtllll, 2023-03-12:
Hi professor fossen
In your book **Handbook of Marine Craft Hydrodynamics and Motion Control (2011) ** I have some questions
1. P125 equation (6.76), (6.77) and (6.81), the coefficient maybe should be $2 \pi$ not $8 \pi$ ?
2. P125 equation (6.78)-(6.80), the coefficient 2 maybe should delete?
3. P126 figure 6.4, the second figure's legend equation should change into $C^{new}_f=(A_x/S)C_x$ or something. The equation $C^{new}_f=(A_x/S)C^{new}_f$ doesn't make any sense.
4. P208 equation (8.66), I just want to know how to choose the value of $\delta\omega$ to compute $A_k$?
Looking forward to your reply.

cybergalactic, 2023-03-13:
1. Yes, this is a known typo in the book.
2. The number 2 is coming from (6.70) such that you obtain 2*  zeta * omega
3. Yes, you are right
4. Dividing the wave spectrum into 50 columns is customary. Sometimes 20 works fine. Hence, if you use omega values from 0-2 rad/s to approximate the wave spectrum, we obtain dw = 2/50 = 0.04 for a 50 columns fit.

wtllll, 2023-03-13:
您好，已收到您的邮件，谢谢！

wtllll, 2023-03-13:
Thanks for your reply！I'm still confused about questions 2 and 4:
2. From (6.70), $2\zeta \omega_n=(d+k_d)/m$, and $k_d=m/T=d$, so $2\zeta \omega_n=2d/m$, right? So the coeffecient 2 In equations (6.78)-(6.80) should be cancelled?
4. In the equation (8.66), it seems that the value of $A_k$ is proportional to the value of dw, should there be a dw in the left side such that $1/2 A_k^2 dw=S(w_k)dw$ ? Moreover, when it comes to a short-crested wave how to choose the value of $d \beta$ ?

cybergalactic, 2023-03-13:
2. You are comparing the following two 2nd-order systems
   m dv/dt + d v + k x = 0
   m ( dv/dt + 2 zeta omega v + k x ) = 0

Hence,  d/m = 2 zeta omega.

4. Equation (8.66) is indeed correct. You can check the famous hydrodynamics textbooks of Faltinsen or Newman for details on this.

By the way, my 2023 lecture notes on the Web are more updated on this topic (less typos). I am not sure how to find the best value for dB since I have not coded this myself. This will be trial and failure, depending on your CPU's speed.

wtllll, 2023-03-13:
Thanks! I guess I got it!
2. Does that mean PD control only works in surge,sway and yaw and does not work in heave,pitch and roll?
4. If the dw value is large in (8.66), the value of $A_k$ will be large but the number N in the equation (8.67)   $\sum\limits_{k = 1}^N $ will be small and vise versa. So the sum value in equation (8.67) will keep almost unchanged. Am I right?
I have read the  lecture notes, but still have trouble understanding these equations.

cybergalactic, 2023-03-14:
2. For marine craft, we use PD or preferably PID in surge, sway, and yaw. These modes do not have a "physical" spring but the P-gain in the control law adds a "virtual" spring such that the closed-loop system is of order 2. Then you can use pole placement with zeta and omega.
4. The number k = 1,2,...,N are the number of dw you pick. If you dive the wave spectrum into 10 columns you get 10 k-values, one for each column. You compute one A_k for each column (k-value). In (8.67) you sum up your 10 columns using k = 1,2,...,10.

wtllll, 2023-03-14:
Ok, I think I fully understand these questions, thanks a lot for your patience answer!

closed 2023-03-14 by wtllll