cybergalactic/MSS #34: Model Otter has a pitch angle that is not 0.

JasonWang2333, 2023-03-27:
Hi there , professor Fossen, i have a problem that the model Otter has a itch angle that is not converged to 0，even its thrust power is se t to zero, is something went wrong or is just normal?

cybergalactic, 2023-03-27:
This is due to the payload. You can change the mass distribution to obtain zero pitch, but the vehicle usually operates with a non-zero pitch angle caused by the payload.

JasonWang2333, 2023-03-28:
> This is due to the payload. You can change the mass distribution to obtain zero pitch, but the vehicle usually operates with a non-zero pitch angle caused by the payload.

Thanks for replying me ,professor Fossen. But i still have another problem in this simulation: the heave angle is not converged to zero but a small constant negative value (around -0.3) ,which may mean that the vessel is going up or down, obviously it is strange, so what is the reason behind that ?

cybergalactic, 2023-03-28:
I can see that this is confusing. The idea is that changing the payload mass and location will change the steady-state heave position, roll angle, and pitch angle. I have updated the SIMotter.m and otter.m files on GitHub to make these more intuitive. In the SIMotter.m file, you can specify a non-zero payload (load condition) by: 

% Load condition
mp = 25;                       % payload mass (kg), max value 45 kg
rp = [0.05 0 -0.35]';     % location of payload (m)

This will result in a non-zero steady-state heave position and pitch angle. If you remove the payload mass

mp = 0;

all the states converge to zero except the roll angle (the propeller generates a non-zero roll moment).

JasonWang2333, 2023-03-28:
Thanks a lot, i will examine the new version and see the results later. Thanks again for your on-time reply.

closed 2023-03-28 by cybergalactic