cybergalactic/MSS #36: The orgin point of the "Otter. m"is not clear

JasonWang2333, 2023-04-11:
Sorry for bothering again, professor Fossen.
As far as I know , the orgin point of a vessel is usually set in the center of gravity or the center of the water plane, but the rp is set as' rg = [0.2 0 -0.2]'; % CG for hull only (m)'. So I wonder where it is set  since i don not the exact data even in the product introduction in www.maritimerobotics.com. It is siginificant cause I want to know some other coordinates in the vessel. Thanks.

cybergalactic, 2023-04-11:
In my 2021 textbook,  CO is the user-specified coordinate origin of the body-fixed frame. CO represents the coordinate origin of the guidance, navigation and motion control systems. We usually chose CO midships on the centerline such that z=0 is the waterline. Note that CB and CG can be time varying. Hence, it is convenient to specify CB and CG with respect to CO using the lever arms rg and rb.

cybergalactic, 2023-04-11:
If you simulate the otter.m file with mp = 0 (zero payload), it will compute the draft corresponding to two pontoons according to

nabla = (m+mp) / rho;         
T = nabla / (2 * Cb_pont * B_pont * L)         // You need to uncomment this line in otter.m
T =
    0.1951

This will be the draft when you have a zero payload. This also gives z = 0, which is my CO. I get a new nabla value if I add a payload mp = 25 kg or mp = 45 kg. This gives T = 0.1951 m and T = 0.2439 m, respectively. Since I add the payload as an external force vector g_0 in otter.m this will change the steady-state z values by approximately 4 and 11 cm since the vehicle is pressed downwards in the water. Removing g_0 from the code will give you z = 0 again, but with a new draft and a new CO.

I am skeptical of the 38 cm value in the Otter brochure you are referring to. This might be the height of the pontoon and not the draft.  If this value is correct (as a draft), my pontoon block coefficient Cb_pont = 0.4 is wrong. For instance, reducing Cb_pont by a factor of 2 will increase the draft from 19.51 cm to 39.02 cm. I need to check the vehicle in the lab to confirm that 19 cm is the physical draft and not 38 cm. NB! There have been different versions of the Otter over the years, so my vehicle might differ from the one in the brochure. 

To conclude:

- You can always get z = 0 by multiplying the payload force g_0 by zero in the m-file. This changes your CO.
- You can get whatever draft you like by adjusting the block coefficient Cb_pont. This changes the buoyancy force of the pontoons.

closed 2023-04-12 by cybergalactic