cybergalactic/MSS #38: About the Hoerner function in the simulation of Otter.m

JasonWang2333, 2023-08-21:
Dear professor, when i compute the cross-flow drag in the otter.m while using the 'crossFlowDrag' function, i found something uncommon, that is:
'
rho = 1025;             % density of water
n = 20;                 % number of strips

dx = L/n;             
Cd_2D = Hoerner(B,T);   % 2D drag coefficient based on  Hoerner's curve
'
in this part, we need to calculate the result of Cd_2D, which is using the Hoerner function to interpolate the Hoerner curve to get the Cd_2D. However, in this part, if we do not apply payload mass, then the B  and T (which i have calculted) are:
'B=1.08(m), T=11/82=0.1341463415(m)'
so that we can get B/(2T)=4.0254545444>4.00309.
but in the function Hoerner ,we only get：
‘% CD_DATA = [B/2T  C_D]
CD_DATA = [...    
0.0108623 1.96608
0.176606 1.96573
0.353025 1.89756
0.451863 1.78718
0.472838 1.58374
0.492877 1.27862
0.493252 1.21082
0.558473 1.08356
0.646401 0.998631
0.833589 0.87959
0.988002 0.828415
1.30807 0.759941
1.63918 0.691442
1.85998 0.657076
2.31288 0.630693
2.59998 0.596186
3.00877 0.586846
3.45075 0.585909
3.7379 0.559877
**4.00309** 0.559315];’
which may get a NaN result when calculating, and i have examined the original reference paper, and i found the curve hasn't the part that over 4.00309, so i wonder how to solve this dilemma while using the 'Hoerner' function.
thanks!

wtllll, 2023-08-21:
您好，已收到您的邮件，谢谢！

cybergalactic, 2023-08-21:
A quick fix is to extrapolate the curve. Given the shape of the curve, I have chosen an asymptotical value by adding the following code to Hoerner.m,

if B/(2*T) <= 4.00309
    CY_2D = interp1(CD_DATA(:,1),CD_DATA(:,2),B/(2*T));
else
    CY_2D = 0.559315;
end 

Hence, the function interpolates up to B/(2*T) = 4.00309, and for larger values of B/(2*T), we use the value  CY_2D = 0.559315.

[image]

JasonWang2333, 2023-08-24:
thank you, i got it.