cybergalactic/MSS #39: Branch(es) to support previous releases of The MathWorks toolchain

mvschaefer, 2023-08-24:
Dr. Fossen,

Thanks for the repository.

FYI, I used portions of this over 10 years ago and am thrilled to see it is still actively being developed!

I have some project needs that require prior release (i.e., r2015a, r2018b, etc.).

FWIW, I have written scripts to search the directory structure, identify mdl/slx models/libraries, do a load/save as, etc.

If it will be of use, I can provide some scripts for you.  We can iterate as necessary to get to a point where a clean checkout of a branch works for my purposes (e.g., r2015a, r2018b) and that should give some assurances that other TMW release branches should be easy to facilitate.

With gratitude.

Mike Schaefer

cybergalactic, 2023-08-24:
Backward compatibility is always an issue.  Are there specific Simulink files or m-files that you need? Converting all the Simulink code to an old Matlab version might be time-consuming. Notice that In the Simulink Editor, you can select
 "Save > Previous Version" on the Simulation tab. Most m-files will probably work out of the box.

closed 2023-09-10 by cybergalactic

mvschaefer, 2023-09-19:
Dr. Fossen,

Apologies for not responding sooner.

I've attached an m-script (saved as text) that should work or be _very_ close to providing automated support for prior releases to whatever level The MathWorks supports slupdate()...

If you can make a branch for R2021b, that 
would be appreciated.  If there are any issues w/the script, I'll be happy to assist in updating it to R2023.

Mike Schaefer

[migrate_libs_m.txt](https://github.com/cybergalactic/MSS/files/12665618/migrate_libs_m.txt)