cybergalactic/MSS #40: Ability to simulate a manipulator attached to a USV

IamGyal, 2023-08-29:
Hi Prof. Fossen,

Would it be possible to simulate a serial manipulator attached underneath a USV with this library in such a way that the hydrodynamics is applied to the manipulator as well?

Thank you very much.

cybergalactic, 2023-08-29:
The toolbox does not contain multibody systems. However, you can use the Otter USV model (scale it up and down) and add a robot manipulator model. Then, you need to understand the interaction forces and moments between the two rigid bodies. Concerning hydrodynamic added mass and damping for a robotic manipulator, publications on the Internet describe these terms. However,  these forces are minor in practice, so a practical approach is to use a standard rigid-body model for the robot manipulator. Since it is submerged, you should still add buoyance forces corresponding to the displaced water volume of the manipulator.

IamGyal, 2023-09-05:
> The toolbox does not contain multibody systems. However, you can use the Otter USV model (scale it up and down) and add a robot manipulator model. Then, you need to understand the interaction forces and moments between the two rigid bodies. 

Thank you very much for your reply.

Would you mean that I need to code the calculation of the interaction forces and moments between the Otter and the manipulator?

cybergalactic, 2023-09-05:
Yes, you need to calculate the interaction forces using other tools/programs.

closed 2023-09-10 by cybergalactic

IamGyal, 2023-09-14:
Thank you very much, Professor!