cybergalactic/MSS #42: Accuracy of simulation results

VastSeaL, 2023-09-19:
Hi Prof. Fossen,

I want to use the motion data, such as velocity、acceleration and position, which caculated from the "otter.m". I'm wondering these results  are accurate enough to show the otter USV navigation in the reat environment.

Thank you very much.

cybergalactic, 2023-09-19:
Yes, this should work nicely.

closed 2023-09-26 by cybergalactic

VastSeaL, 2023-09-28:
Hi Prof. Fossen,
Weather the "otter.m"  calculate the motion state is navigation in the calm water with only considering the current. Can I use the MSS toolbox to simulate the otter motion state in different sea conditon.

cybergalactic, 2023-09-28:
If you want to simulate ocean waves, you need to add the wave loads yourself. This is described in Chapter 10 in my textbook/lecture notes. Note that the Otter USV is a small vehicle so it will not be able to handle large waves.