cybergalactic/MSS #43: about 'otter.m'

JasonWang2333, 2023-10-14:
Hi there，professor Fossen, I found that the motion equation of otter was  using the function of follows:
[image]
so i got 2 questions about it , because i wanna use it to simulate and finish my paper.
The FIRST is, in your book , it says:
[image]
therefore, could i say that, the model you use in the otter.m can only be used in motion of line motion? what if the USV follows a curve? Can i use it to follow the Polyline path？Or i have to use the following equation in your book : 
[image]
and reconstruct the model?
The SECOND is , i didn't know much about the crossflow drag and strip theory:
[image]
but i had looked into some paper and found the strip theory is often used in the calculation of wave disturbance, so i wonder if the crossflow drag the same as wave disturbance? Also, i didn't know where the origin of the equations here:
[image]
though i know the Cd_2d coefficient is based on a curve of a paper.
Besides, i have some other questions about the model,  can you reply me the answer even though i know your time is limited and precious?
THIRD, you know the otter model is a 6DOF model, but usually we control the model of 3 degree, so it needs to simplify the model  and transform it into 3 DOF, but i wonder how to decouple the 6DOF model into 3DOF, the only way i could do is to delete the rows and columns of w\p\q ， i wonder if there is better way to do that?
FORTH,  about the hydrodynamic coefficients in this model.
i know all of the hydrodynamic coefficients you use have their experiments and statistics support, however, if i have to consider the variable mass of mp,which may change the draft , as a result , the  hydrodynamic coefficients as well as the draft may be variable all the time, so i wonder if these equations effective while the mass is changing?If it is not  effective, how can i consider the hydrodynamic coefficients of otter?
[image]
[image]

Thanks !

wtllll, 2023-10-14:
您好，已收到您的邮件，谢谢！

cybergalactic, 2023-10-14:
1) The otter.m model is a 6-DOF model, valid for all motions (not only straight lines). Your question regarding r = 0 is related to ocean currents. The model uses a horizontal ocean current with velocities uc and vc. There are no vertical currents wc. In addition, it is assumed that the time derivative of the ocean current velocity expressed in BODY is zero. This means that the S(omega) v_c term in

[image]

is assumed to be zero (only valid if r = 0 or small). You can remove this assumption by adding the S(w) v_c term in the model.  In practice, this term will be minimal. Note the Otter USV model is not affected by this. It is only the current model. However, I will add this term to the GitHub model since you are among many asking this question. 

2) Wave disturbances are not related to the cross-flow drag formula. Strip theory means you slide the hull into pieces and use a 2-D formula for each piece. Then, you sum up along the x-axis (like an integral).  This is the reason you need the 2-D Hoerner formula for drag. The cross-flow drag formula is a nonlinear damping force in sway and yaw using the relative velocity (including the currents' contribution). It is not a wave force. The nonlinear damping terms are in addition to the linear damping terms.

3) You should not simplify your 6-DOF equations of motion. The model should be a high-fidelity 6-DOF model representation of the actual system. Instead, you should make a measurement model, selecting the states you need. You choose the measurements corresponding to surge, sway, and yaw. Then, you compute your control law for the same 3 DOFs. Finally, you define a 6-DOF vector for the control input and substitute your three controls into the vector's surge, sway, and yaw positions. The remaining three elements should be zero. Consequently, there is no need to decouple or reduce the model.

4) You have a new model if you change the load condition (mass, payload, etc.). Then, all mass parameters, CG, CB, and added mass terms change. You can interpolate or extrapolate the data; alternatively, do all computations from scratch for each load condition.

JasonWang2333, 2023-10-14:
thanks a lot!

cybergalactic, 2023-10-14:
The otter.m has been updated on GitHub. It now includes the missing S(w) v_c term. The file ExOtter.m explains how to design a course autopilot using the 6-DOF model. No need for model reduction.

JasonWang2333, 2023-10-15:
I see.
And I will see the updated codes immediately and update my simulation model as well. 
Thanks for your kindness and patience.

closed 2023-10-15 by cybergalactic