cybergalactic/MSS #47: Can I use the Remus100.m file to perform zigzag and turn circle maneuvers on a horizontal plane?

wanghengshine, 2023-11-17:
Dear Professor Fossen：

I want to study the characteristics of horizontal motion of the Remus100 underwater vehicle. Can I use the Remus100 file to perform zigzag and turn circle simulation experiments?
If possible, how should I modify the existing ship model file? Could you provide some guidance or ideas?

Best Regards
Heng Wang

wtllll, 2023-11-17:
您好，已收到您的邮件，谢谢！

cybergalactic, 2023-11-18:
You can modify the example file ExZigZag.m to call the remus100.m model file. The file is a template file used to call a ship model.

wanghengshine, 2023-11-18:
Dear Professor：
I attempted to modify the ExZigzag file, with the initial conditions set as follows：
`x     = [1.25 0 0 0 0 0 0 0 0 0 0 0]';
delta_r = 0;
delta_s = 0;
n       = 750;`
the graph of lateral velocity v versus time t is as follows
[image]
Why does the longitudinal velocity  fluctuate frequently? Is it a problem with my code debugging?

cybergalactic, 2023-11-18:
It can be many reasons. The sampling time may be too high. Try 50-100 Hz. Turn off ocean currents?

wanghengshine, 2023-11-18:
I need more data, so I set `t_final = 100; t_rudderexecute = 2; h = 0.1;`and I turned off the ocean currents.
The graph is from the 10° zigzag test I modified, but I can't find the reason for the fluctuations in the longitudinal velocity.

cybergalactic, 2023-11-18:
I have added the Remus 100 AUV to the ExZigZag.m file for your convenience. You need to download the updated ExZigZag.m and zigzag.m files from GitHUb. In addition, there is a new file zigzag6dof.m for 6-DOF models such as the Remus 100. Running the new example file gives you the Remus 100 zigzag plots shown below.

[image]

[image]

wanghengshine, 2023-11-18:
I am very grateful for Professor Fossen's help, and I will try to calculate the data again.

closed 2023-11-19 by cybergalactic