cybergalactic/MSS #48: Otter in simulink

VastSeaL, 2023-12-11:
Dear Professor Fossen：

When I use the Demo of otter in simulink, I was confused in how to calculate revolutions. Firstly, in Control allocation Function, why the output is 'abs(n)*n' and square root subsequently, and why not output 'n' directly, in FIgure1. And in figure.2, what's the meaning of 1/(s+1).

Best Regards
Luo
[image]
[image]

wtllll, 2023-12-11:
您好，已收到您的邮件，谢谢！

VastSeaL, 2023-12-11:
[image]
Sorry, Professor Fossen. This is Figure 1.

cybergalactic, 2023-12-11:
The transfer function 1/(s+1) is an optional low-pass filter used to model the DC motor dynamics. You cannot obtain a propeller speed command in 0 seconds. It will take some time. The time constant is chosen as 1 second in the model, but you can change it if you like.

Control allocation can be solved using dynamic optimization. However, this is complicated and time-consuming. Hence, I inverted the static mapping from propeller speed to force. This is a quadratic function with a sign, so you must be careful. Note that

    tau =  |n| n

The inverse mapping is 

    n = sign(tau) * sqrt( |tau| )

In addition to this, you need to invert the input matrix to get the correct input gains in sway and yaw. This is a 2x2 matrix where the number 0.395 is defined in otter.m. You need to study the control allocation chapter in my textbook/lecture notes to understand how I choose the 2x2 matrix. Alternatively, you can design your own control allocation algorithm.

closed 2023-12-14 by cybergalactic