cybergalactic/MSS #51: Question about the difference between zero-frequency and infinite-frequency added mass matrix

wtllll, 2024-01-28:
Hi @cybergalactic,
I'm really confused about the physics meaning of the zero-frequency added mass matrix $A(0)$ and infinite-frequency added mass matrix$A(\infty)$ . Also, the physics meaning of the zero-frequency potential damping $B(0)$ and infinite-frequency potential damping $B(\infty))$.

In my imagination, the zero-frequency added mass matrix $A(0)$ and the zero-frequency potential damping $B(0)$ are the added mass matrix and potential damping when the vehicle in calm water, right? Then, what's the physics meaning of the infinite-frequency?

Best regards

cybergalactic, 2024-01-28:
It might help to think about this by considering a regular wave, z = A * sin(omega * t), exciting your vessel. When you do a physical experiment, generating a wave with infinite frequency is impossible. In practice, the waves will have a positive nonzero frequency omega.  The zero frequency gives z = 0, and there are no incoming waves. The infinity frequency is a non-physical value. It is the asymptotic value and the waves can never reach the infinity frequency. However, they can have a very high frequency close to the infinity value.

wtllll, 2024-01-28:
您好，已收到您的邮件，谢谢！

wtllll, 2024-01-28:
Got it! Thanks for your explanations!

closed 2024-01-28 by wtllll