cybergalactic/MSS #52: Vertical velocity used in cross-flow drag calculation for AUVs

gabrielfpacheco, 2024-01-29:
Hi Prof. Fossen,

I noticed that in [this commit](https://github.com/cybergalactic/MSS/commit/c3b4ab55b07e490c5592a58845ec4c28fc25f97d) you included heave force and pitch moment to `crossFlowDrag` which are relevant for AUVs. Regarding this point, I have two questions:

1. Shouldn't [U_v](https://github.com/cybergalactic/MSS/blob/72c0e278a3aee329b7b1d430d411e0e31a3b70dc/GNC/crossFlowDrag.m#L30) use the heave relative velocity `w_r = nu_r(3)` and pitch rate `q = nu_r(5)`, i.e.: `U_v = abs(w_r + xL * q) * (w_r + xL * q);`, or am I missing something? 

2. Considering that a low aspect-ratio wing approximation that takes the vehicle angle of attack as an input is being used to compute lift/drag/pitch moment for the hull. Would such effects (extra heave force and pitch moment) already be accounted for, or do they have a different nature?

Thank you in advance.
Best regards.

cybergalactic, 2024-01-30:
Thanks for sharing this.

1) You are right, it should be U_v = abs(w_r + xL * q) * (w_r + xL * q).

2) A torpedo-shaped UAV will not experience lift at low speed. At higher speeds, the lift will force the AUV upwards, and you need to account for this by using the control surfaces and/or a  ballast system. The AUV model does not include any compensation for the lift force, so you need to design an autopilot with integral action.  This will produce the necessary action to counteract the lift force. It will be observed as bias in the control surface (elevator) deflection.

gabrielfpacheco, 2024-01-30:
1. Thank you for the [fix](https://github.com/cybergalactic/MSS/commit/c19baed723bd395d5dba60a860065079bc941dda), Prof. Fossen.

2. I apologize if my previous question was unclear. My concern is not regarding the control action required to compensate for unmodeled lift effects. Instead, I am more focused on ensuring an accurate simulation of the nonlinear damping effects, while avoiding any duplication of the same phenomenon.

> The AUV model does not include any compensation for the lift force

However, if I understood your explanation correctly, the cross-flow drag on heave and pitch would be this missing part to dampen the effects from lift (and eventually pitch moment) that become more representative in higher speeds.

cybergalactic, 2024-01-31:
The model has quadratic velocity terms due to Coriolis and centripetal forces. These are caused by the rotation of the BODY frame about the NED frame, which acts as an (approximative) inertial frame. In addition, the model has nonlinear damping terms computed from the cross-flow drag principle and the lift-drag forces of a low aspect-ratio wing. 

The cross-flow drag method computes a sway force and yaw moment in the horizontal plane by integrating the 2-D drag coefficient along the axis. This is the classical strip theory used for ships.  Since the AUV is modeled as a cylinder, the same principle can be used to compute the force and moment in the vertical plane to obtain the heave force and pitch moment. However, this is a decoupled approach (using two planes) since the forces and moments acting on an AUV will be coupled in the 3-D case.  Decoupling can be justified for "slender bodies" moving at high speed. For most AUVs moving at 2-3 m/s these terms will be very small and even neglected.

gabrielfpacheco, 2024-02-01:
Thank you so much for the explanation/clarification, Prof. Fossen. It was very helpful.