cybergalactic/MSS #54: about the mass and draft

JasonWang2333, 2024-03-17:
Dear professor Fossen. I have a question about the linear speed. You know, I used to discuss the model 'Otter.m' for quite a few times, however, I found a question about the payload mass 'mp', that is , when I input the mp, which is not zero, then the  ship began to have a heave speed which is not converged to 0, also ,when I increase the mp,the heave speed began to decrease and the surge speed increased, and I don't really know the reason, even if I check the codes for a while and the text book you wrote to check the equations ,i still dont find the answer. ps: the model is created based on ‘otter.m' file.Here is the simulation results:
[image]
in the past, I don't consider the payload mass, so i miss that porblem, that is the reason why i come here again.

JasonWang2333, 2024-03-17:
Also,even in 'otter.m' as well as a'SIMotter.m', it has the problem metioned above:
①mp=0kg, the 'Heave Velocity≈0m/s'
[image]
②mp=20kg, the 'Heave Velocity≈-0.05m/s'
[image]
I wonder that if it is the problem of lack of resistance compensation in the direction of buoyancy.

JasonWang2333, 2024-03-18:
Well, thanks again for your kind and efficient reply,i run the new simotter and it seems that the heave-velocity problem is solved. However, As i mentioned above, i still found the surge speed seems strange, you know, when  the payload mass increases, the draft increases so that the drag increases, so the USV is supposed to run slower than with the smaller payload( if the rotation rate of the propellers are set as constants). even when i set the payload mass as zero,namely,mp=0, the surge speed is the minimum among them.:
in the first pic,(the n1=n2 which are constants, the payload is set as 0kg,the heave and sway speed are around 0.)
[image]
in the second pic, the payload is set as 10kg, other settings are the same with the first one.
[image]
you can see that the max surge speed is increased from 1.426m/s  to 1.528m/s
pic 3rd is the scope of the simulation
[image]

cybergalactic, 2024-03-18:
The current model does not consider that the angle of attack is nonzero when you trim the vehicle. If this is important, you must add a more sophisticated resistance function in surge. For the moment, linear damping in surge is computed by specifying Umax as the constant value corresponding to n1 = n2 = n_max

  Umax = 6 * 0.5144;             % max forward speed (m/s)
  Xu = -24.4 * g / Umax;        % specified using the maximum speed , the number 24.4 comes from the propeller experiment

corresponding to

   (m - Xudot) * udot - Xu * u =  k_pos * n1 * abs(n1) + k_pos * n2 * abs(n2)

In steady state udot  = 0, and you can compute Xu using  u = Umax and n1 = n2 = n_max such that 

   Xu = - k_pos * (n_max^2 + n_max^2 ) / U_max
  
This formula does not depend on the angle of attack. The simple update is to define U_max as a function of the pitch angle to obtain a speed reduction for large pitch angles.  A more advanced approach will be to compute the friction coefficient as a function of the angle of attack and include formulas for surge damping using the friction coefficient.

JasonWang2333, 2024-03-18:
well, i got ur idea, i will try it later. thanks a lot!

closed 2024-03-18 by JasonWang2333

JasonWang2333, 2024-03-19:
Professor Fossen, I have tried to use:
Umax = 2.572*(2*eta(5)+1);
Xu = -24.4 * g / Umax;        
to calculate the payload question, and the results seem like this
[image]
in which the surge velocity and distance of x seems to decrease when the payload increases
[image]
[image]
pic1: 2kg payload,139.5m,1.426m/s
pic2: 3kg payload,137.5m,1.405m/s
pic3: 10kg payload,129.9m,1.327m/s

it seems quite reasonable , but i wonder:、
① if it is proper and accurate to use the 
Umax = 2.572*(2*eta(5)+1);
to compensate for the nonzero pitch angle, and what is the result of not accurate coefficients of eta(5)?  
②
also， I wonder if is OK to use this model to simulate the motion while the ship has a nonzero but small roll angle?（like-0.9°）

JasonWang2333, 2024-03-19:
@cybergalactic

cybergalactic, 2024-03-19:
Scaling the maximum speed so that it corresponds to the maximum thrust seems reasonable, even for varying pitch angles. You need speed, pitch angle, and RPM time-series measurements from the real vehicle to improve the results further.  Experimental calibration using a time series is always preferred since the simulator is a simplified model based on approximation formulas. 

Concerning the speed reduction for a small roll angle induced by moving the payload sideways, this will not cause a significant change in speed as long as the roll angle is small. -0.9 degrees is indeed a small angle, and the frontal submerged area of the vehicle nor the drag coefficient will most likely not change significantly.

JasonWang2333, 2024-03-20:
Well, I get it, thank you!

cybergalactic, 2024-03-28:
I have done many new updates to the otter.m and SIMotter.m files. The problem with the nonzero heave velocity was due to the trim condition, not the numerical integration method in SIMotter.m.

1)   In the updated otter.m file, the trim condition is now computed using the equilibrium state. This seems to solve the problem.
2) The new SIMotter.m has many upgrades including:
- Numerical integration is done by using forward Euler and the xdot value computed in otter.m 
- Path following and heading control are included. You can change this with a flag
-  The reference model is called as a function
- Improved plotting
-  and other fixes