cybergalactic/MSS #56: I would like to know if there are any related video tutorials available for learning.

trouvaillezhao, 2024-03-25:
Hello. I am interested in the ' Handbook of Marine Craft Hydrodynamics and Motion Control ' course you mentioned on GitHub. I want to know whether there are relevant video tutorials to learn ? I have seen detailed information about the course in the course introduction, but I want to further understand whether there are video materials available for learning. Thank you for your time and help !

cybergalactic, 2024-03-25:
I am sorry, but there are no videos available. The only materials online are the lecture notes and the software.

trouvaillezhao, 2024-03-25:
Thank you for your response. I appreciate your time in providing clarification. While it's unfortunate that there are no video tutorials available, I'll make sure to utilize the lecture notes and software resources effectively. Thank you again for your assistance.

closed 2024-03-26 by cybergalactic