cybergalactic/MSS #58: issue about the ILOS guidance law based on the heading(psi) and course angle(chi)

JasonWang2333, 2024-03-26:
1.in file'ILOSchi'&'LOSchi', the y_e expression is written as follows:
[image]
however, the chi contains beta_c, which is compensated by the integration item 'y_int'：
[image]
so i think it should be written  as:
[image]
but i run the new ILOSchi, and it works worse than the former one.(maybe because of  the estimated y_e)

2. I used to use the ILOS guidance law to guide my USV, in which  I used chi_d as the desired angle to lead the ship. I found that in your  article  ’An Adaptive Line-of-Sight (ALOS) Guidance Law for Path Following of Aircraft and Marine Craft‘, an ALOS method was introduced at first, but it used the psi angle to lead the ship and use adaptive method to compensate for the crab angel 'beta_c', and i checked it out   and found the heading and course angle method are similar other than the former one uses the psi angle and the y_e expression uses U instead of Delta:
[image]
[image]
Is it OK to change the guidance law into psi-leading just by changing the equations?
3. I read your text book about the LOS, I noticed that the crab angle is always set to unknown,  so the methods of ILOS AILOS ALOS etc. are methods to compensate for the crab angle in theta_1 or theta_2  as follows:
[image]
so if I can read the  u and v velocities , is there any need to compensate for the beta_c?(in the wind wave and flow environment)

cybergalactic, 2024-03-27:
The very important proof that 

    psi = chi - beta_c

is derived using the amplitude-phase representation, which I managed to derive in 2023 and publish in the paper

Fossen, T. I. (2023). An Amplitude-Phase Representation of the North-East-Down Kinematic Differential Equations. IEEE Access 11, 12586-12592.  https://doi.org/10.1109/ACCESS.2023.3242331](https://doi.org/10.1109/ACCESS.2023.3242331

The consequence is that the cross-track error expressed in the path-tangential frame satisfy

   dy_e/dt = U * sin(chi - pi_h) = U * sin(psi + beta_c -pi_h) 

This gives you two choices for path-following design:

1. If you measure psi (compass), you must compensate beta_c by integral action or direct measurements. 
3. No compensation of beta_c is needed if you measure (or compute) the course angle chi. 

The older version of the MSS toolbox has a file ILOSchi.m. The file is removed in the current version of the toolbox (only LOSchi.m exists) since you DO NOT need integral action when controlling the course angle chi. The integral term will not improve the path-following capability and is very hard to tune. However, integral action is needed when you use the compass angle psi to reduce the cross-track error. The corresponding MSS file is ILOSpsi.m.

Yes, you can compute beta_c from direct measurements since beta_c = atan(v / u).  This is tempting, but the value of beta_c is numerically ill-conditioned for small u and v, which is common for marine vehicles. You obtain much better performance by using ILOS or ALOS (ILOSpsi.m and ALOSpsi.m), which treats beta_c as a nearly constant drift term.  Note that beta_c is not defined at zero speed, and this makes compensation by measurements hard.

JasonWang2333, 2024-03-28:
thanks a lot.  And i found the new file 'ILOSpsi.m' has a problem:
 r_d = -Kp * Dy_e / ( 1 + (Kp * y_e)^2 );      % d/dt psi_d
because here uses integration method, it  should be :
D_int=Delta * y_e / ( Delta^2 + (y_e + kappa * y_int)^2 );
r_d =-Kp*(Dy_e+kappa*D_int)/(1+(Kp * y_e+kappa*y_int)^2);  % d/dt psi_d
y_int = y_int + h *D_int ;

② and I wonder what is this transfer function or delay for？
[image]

cybergalactic, 2024-03-28:
The desired yaw rate r_d is computed assuming y_int = 0. As far as I remember, there was no improvement in adding the y_int term; it only caused stability problems. You should test this yourself.  Anyway, the additional y_int term is unnecessary since I use an observer to compute the filtered psi_ref value.

% observer/filter: psi_f[k+1]
psi_f = psi_f + h * ( r_d + K_f * ssa( psi_ref - psi_f ) )

If r_d is a little wrong here, this still works thanks to the dominant feedback term K_f * ssa( psi_ref—psi_f ), forcing psi_f to track psi_ref.

This "hack" and filtering of psi_ref are due to the switching of waypoints, which causes jumps in the control signals. A better approach would be to use Hermite spline interpolation through the waypoints. Then all jumps go away. I will add code for this later.  Hard switching of waypoints should be avoided in industrial implementations.

The Simulink diagram's low-pass filter, h(s) = 1 / (Ts+1), is included for the same purpose. It avoids jumps in the reference signal r_d due to waypoint switching. If you use a smooth Hermite spline, this is not necessary. Again, this will be updated in the future.

JasonWang2333, 2024-03-28:
well I get it. thanks a lot

closed 2024-04-08 by JasonWang2333

cybergalactic, 2025-05-21:
The autopilot uses ssa(psi-psi_d) to map the tracking error e=psi-psi_d to [-pi pi].  This should be applied to the P and I terms in the control law. You do not map the D term. Hence, this is a safe implementation of the PID autopilot..The LOS guidance law should use atan2 such that you get a command in the interval  [-pi pi]. Then your desired signal should follow the command without any jumps.

renjungen, 2025-05-21:
[image] 

[image]
Thank you for your reply! I am using atan2, and my controller is ADRC. I have mapped the actual heading angle, LOS angle, and error angle to the range [-π, π]. I define the north direction as 0 degrees, with the angle increasing clockwise. As shown in Figures 1 and 2, I added external disturbances, and the desired heading angle oscillates between -180 and 180 degrees. However, my actual angle cannot track it well. Is this because my controller is not well-designed, or should I not worry about this situation for now? Since this is for my master's thesis, if I use waypoints with a slope, I can achieve good tracking performance.

renjungen, 2025-05-21:
Do I only need to map the error angle to [-π, π], while the actual angle and LOS angle don’t have to be constrained within -180 to 180 degrees? I’m quite confused about this. Thank you for your explanation!"

cybergalactic, 2025-05-21:
The control law can be too aggressive and this will give you oscillations. Note that Kp = 1 / Delta in the LOS guidance law. Large Delta reduces the proportional gain. Also note that you have two control loops. The inner-loop is a heading controller and the outer loop is the guidance law. Hence, you need bandwidth separation. The inner loop should be 5-10 times faster than outer loop.

You need to be careful with your waypoints. The turn you are showing in the plot is hard. Maybe add one or two waypoints. Spacing of waypoints can also be an issue.

Yes, only the error since you want to avoid jumps in the control signal. Heading angle can be 0 to 2 pi or even N times 2 pi.

- Thor

renjungen, 2025-05-21:
Thank you for your answer! Another question: If I perform cubic interpolation on these independent waypoints, is this considered the same type of problem as curve tracking? Can I treat 'cubic interpolation tracking of waypoints' and 'curve tracking' as two separate chapters in my thesis?"

cybergalactic, 2025-05-21:
Yes

Yes, you can have two chapters

Straight-Line Path Following via Waypoints
Curved Path Following via Waypoints

The complication is to compute the tangent of your curved path.  For a straight line this is the slope of the curve.

- Thor