cybergalactic/MSS #59: Why is M_A in 'Otter.m'  a diagonal matrix？

JasonWang2333, 2024-05-10:
but in ur text book, 'the vehicle also has three planes of symmetry', when the M_A is a diagonal matrix. 
[image]
 and i check the USV doesn't has three planes of symmetry:
[image]
what's the reason of simplifying this matrix?

cybergalactic, 2024-05-11:
The process of purchasing a hydrodynamic program and creating a 3D drawing for hydrodynamic computations is both time-consuming and costly. Alternatively, you could compute the 6x6 added mass matrix using these methods, or you might opt for simplified semi-empirical approaches. For surface vehicles like the Otter USV, there is port-starboard symmetry. The submerged hull exhibits nearly perfect fore-aft and top-bottom symmetry, with the exception of appendages like propellers. Therefore, it's feasible to model this as having 3-plane symmetry and utilize a diagonal MA matrix, where the off-diagonal elements are significantly smaller than the diagonal ones.

JasonWang2333, 2024-05-21:
Thanks a lot. I get it.

closed 2024-05-21 by JasonWang2333