cybergalactic/MSS #60: possibility to add simulation model for paper "Line‑of‑sight path‑following control utilizing an extended Kalman filter for estimation of speed and course over ground from GNSS positions"

LiangZhao13, 2024-05-12:
Dear professor Fossen, I recently read your publication "Line‑of‑sight path‑following control utilizing an extended Kalman filter for estimation of speed and course over ground from GNSS positions" in JMST. The paper described a complete and detailed model of the application for GNC system. I was wondering is it possible to add the simulation model in the paper to the MSS toolbox in the next release?

cybergalactic, 2024-05-15:
I have included the SImulink file "demoMarinerPathFollowingCourseControl.slx"  representing the case study in JMST. Later, I will also add an m-file version of the case study.

cybergalactic, 2024-05-17:
SIMmariner.m is now available to simulate 2-D LOS path-following control using a course autopilot based on pole placement. The m-file is implemented similarly to the Simulink block diagram "demoMarinerPathFollowingCourseControl. slx".

LiangZhao13, 2024-05-18:
Thank you so much for providing those demos !!!!

closed 2024-05-18 by LiangZhao13