cybergalactic/MSS #61: About the otter usv

VastSeaL, 2024-05-30:
Dear Professor Fossen,
I want to use the otter USV as a object of study, can i know the max current speed for sailing.

cybergalactic, 2024-05-30:
You can find the information in the 'otter.m' file.

% Main data
g   = 9.81;         % acceleration of gravity (m/s^2)
rho = 1025;         % density of water
L = 2.0;            % length (m)
B = 1.08;           % beam (m)
m = 55.0;           % mass (kg)
rg = [0.2 0 -0.2]'; % CG for hull only (m)
R44 = 0.4 * B;      % radii of gyrations (m)
R55 = 0.25 * L;
R66 = 0.25 * L;
T_sway = 1;         % time constant in sway (s)
T_yaw = 1;          % time constant in yaw (s)
Umax = 6 * 0.5144;  % max forward speed (m/s)

Here, Umax is the maximum speed when there are no currents. If you have currents close to 6 knots (approx. 3 m/s), the Otter USV will drift away. In practice, controlling the vehicle will be very hard if the currents are stronger than 1 m/s since you need stability margins. You can test this yourself by running the script 'SIMotter.m' and changing the current speed and direction in the script. The default values are:

V_c = 0.3;                           % Ocean current speed (m/s)
beta_c = deg2rad(30);      % Ocean current direction (rad)

VastSeaL, 2024-06-06:
How to calculate the stability margins that mentioned in your answers? 
Thanks!

cybergalactic, 2024-06-06:
The currents are stochastic, so performing a stability analysis of the vehicle dynamics is not straightforward. In the deterministic case (constant currents), generating a DP capability plot is the standard practice. This plot is obtained by setting the state derivative of the dynamic equation to zero and solving for the steady-state solution when you apply a constant current to the vehicle and, at the same time, apply the maximum thrust.  

https://www.offshoreengineering.com/dp-dynamic-positioning/dp-capability-plot-example/ 

https://www.dnv.com/maritime/services/dp-capability-85124/

In practice, the vehicle must move faster than the ocean currents. You can simulate the vehicle and change the current speed and direction in the m-file. From this, you learn what kind of currents the vehicle can handle.

closed 2024-06-14 by cybergalactic