cybergalactic/MSS #62: On vessel model

LiangZhao13, 2024-06-01:
Dear Professor Fossen, 

I recently find that the inertial/hydrodynamic parameters in container.m are relatively small, while in navalvessel.m are quite large. Is this mean the parameters in container.m are derived from a smaller prototype model?

Thank you

cybergalactic, 2024-06-01:
If you read the reference paper by Son and Nomoto (1982), you will discover that the model parameters in 'container.m' are nondimensional, following the standard convention in naval architecture, see Appendix D in Fossen (2021). The forces and moments are transformed into dimensional quantities at the bottom of the file using L and U. In other words, 

% Dimensional state derivatives  xdot = [ u v r x y psi p phi delta n ]'
detM = m22*m33*m44-m32^2*m44-m42^2*m33;

xdot =[                      X*(U^2/L)/m11
          -((-m33*m44*Y+m32*m44*K+m42*m33*N)/detM)*(U^2/L)
           ((-m42*m33*Y+m32*m42*K+N*m22*m33-N*m32^2)/detM)*(U^2/L^2)
                   (cos(psi)*u-sin(psi)*cos(phi)*v)*U
                   (sin(psi)*u+cos(psi)*cos(phi)*v)*U 
                              cos(phi)*r*(U/L)                
           ((-m32*m44*Y+K*m22*m44-K*m42^2+m32*m42*N)/detM)*(U^2/L^2)
                                p*(U/L)
                              delta_dot 
                                n_dot                 ];

LiangZhao13, 2024-06-02:
Thank you so much for the answer.

closed 2024-06-02 by LiangZhao13