cybergalactic/MSS #66: About CyberShipⅡ

VastSeaL, 2024-08-22:
Dear Professor Fossen,
    When I use the CyberShipⅡ model in MSS toolbox, I would like to know the inertia matrix of this model and other matirx, such as Coriolis matrix. Where can I find in the toolbox?
    Thank you very much for your time and consideration. I understand you have a busy schedule, and I am grateful for any assistance you can provide.

Looking forward to your response.

cybergalactic, 2024-08-22:
This is a linearized model for stationkeeping (low speed). Hence, there are no quadratic damping terms due to Coriolis and damping. Adding quadratic Coriolis and centripetal terms might destabilize the model unless you counterbalance this by adding nonlinear damping (ITTC drag, crossflow drag, etc.). It is straightforward to do this using the methods in my textbook. For instance, the Coriolis matrix is computed using the MSS function: C = m2c(M, nu); see Chapter 3. Damping is explained in the chapter covering maneuvering models.

The mass matrix for the model ship is found in initCS2.m. 

The osv.m and SIMosv.m are alternative and more sophisticated models with Corioilis and nonlinear damping.

VastSeaL, 2024-09-02:
Dear Professor Fossen,
    The CyberShip Ⅱ model ship in simulink, where the inv(M) introduce or input?
[image]

cybergalactic, 2024-09-02:
You will find the model here

…./MSS/VESSELS/models/supply.m

Note that the mass matrix is using bis-scaling.

closed 2024-09-19 by cybergalactic