cybergalactic/MSS #68: The mathematical model of tanker

VastSeaL, 2024-09-21:
Hello, Prof Fossen,
     I want to study the path following control base the tanker model in your MSS toolbox, whrer in tanker.m. But, I do not know the process to caculate the result of "xdot". In other word, I do not understand the equations, as shown in the figure, based on what mathematical model, such as the MMG and Abkowitz model, to get these equations ?
Thank you!
[image]

cybergalactic, 2024-09-22:
The famous paper: 

Van Berlekom, W. B. and T. A. and Goddard (1972). Maneuvering of Large Tankers, Transaction of SNAME, 80:264-298

describes how all the hydrodynamic coefficients are computed. The reference is found in 'tanker.m'.  However, the coefficients are made non-dimensional using the Bis-scaling system in hydrodynamics. I have derived a systematic approach for this in Appendix D, Nondimensional Equations of Motion, in my textbook

Fossen, T. I. (2021). Handbook of Marine Craft Hydrodynamics and Motion Control, 2nd edition, John Wiley & Sons. Ltd., Chichester, UK. ISBN 978-1119575054.

By reading these two references, you should be able to understand the model, including the Bis-scaling system.

closed 2024-09-25 by cybergalactic

VastSeaL, 2024-09-26:
I have understood the model, but why the flow velocity past rudder was chaged shown in Fig. 1. The paper 'THEORY AND OBSERVATIONS ON THE USE OF AMATHEMATICAL MODEL FOR SHIP MANOEUVRING IN DEEP AND CONFINED WATERS' mentioned in "Maneuvering of Large Tankers" use the formula as shown in Fig. 2.
[image]
[image]

cybergalactic, 2024-09-26:
The parameters for the 190 000 dwt Esso Osaka tanker in the Berlekom and Goodhard (1972) paper are listed in Appendix A. There are several models of different tankers in the paper, but for the 190 000 dwt tanker there are only two c-terms, namely cun and cnn.  Based on common practice and how these coefficients are used, the typical formula to compute  c  is:

c = sqrt( cun*u*n + cnn*n^2)

Eq. (7.3)  has four terms and this is a different formula used by other authors. It depends on u and n, but the curve fitting must be different.